IPdotSetAF/ESPAsyncHTTPUpdateServer #2: ESP32 support

IPdotSetAF, 2024-02-24:
add support for esp 32 :)

goebish, 2024-04-21:
I expected it to work on ESP32 as stated in the README but got this error:
The ESPAsyncHTTPUpdateServer library encountered an unknown path resolve error.D*: 2:10: fatal error: flash_hal.h: No such file or directory

IPdotSetAF, 2024-04-21:
> I expected it to work on ESP32 as stated in the README but got this error: The ESPAsyncHTTPUpdateServer library encountered an unknown path resolve error.D*: 2:10: fatal error: flash_hal.h: No such file or directory

sorry, my bad. i was planning to add the support since its not much work but my dev system windows got corrupted. and the issue is left open. 

I will try to add it to the lib and publish it ASAP.

closed 2024-06-04 by IPdotSetAF

IPdotSetAF, 2024-06-04:
Hey @goebish 
Sorry it took some time, I was a little busy.

Could you give it another try and let me know if your issue is fixed or not?