IPdotSetAF/ESPAsyncHTTPUpdateServer #14: Renaming the library

IPdotSetAF, 2024-07-26:
After #13 is done, we need to choose a better name for the library because the `Async` in `ESPAsyncHTTPUpdateServer` does not describe the library completely.

I'm looking forward to any name recommendations anyone has in mind, please leave your ideas in this issue ❤️

IPdotSetAF, 2024-07-26:
I'm thinking of `ShadyOTA` for the shade emoji in the webpage but the name is kinda negative

IPdotSetAF, 2024-08-16:
TinyOTA

IPdotSetAF, 2024-08-16:
SlimOTA

hadoopmarc, 2025-08-01:
How about ESPAnySyncHTTPUpdateServer. This keeps suggesting it works nicely together with other libraries based on ESPAsync. People not needing ESPAsyncWebServer are more likely to arrive at the standard webUpdater example from Espressif, anyway.

marhalloweenvt, 2025-10-08:
I will go with: TinyOTA and ESPsooOTA. I prefer ESPsooOTA more but my mama says that "Don't spill coffee on electronic."