IPdotSetAF/ESPAsyncHTTPUpdateServer #15: Forever Reset After Upload on ESP8266 12 Board ESP8266 Generic

Sebuahhobi, 2024-07-30:
### ESP8266

`
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>

#include <ESPAsyncWebServer.h>
#include <ESPAsyncHTTPUpdateServer.h>
#include <ArduinoJson.h>
#include <Wire.h>
#include <RTClib.h>
#include "PrayerTimes.h"
#include "HJS589.h"
#include <FS.h>
#include <LittleFS.h>
#include "AsyncJson.h"

RTC_DS3231 rtc;
ESPAsyncHTTPUpdateServer updateServer;
AsyncWebServer server(80);

#include "DFRobotDFPlayerMini.h"
DFRobotDFPlayerMini myDFPlayer;

#include "variabel.h"
#include "sh64.h"
#include "helper.h"
#include "jws.h"
#include "fileFS.h"
#include "wifi.h"

#define ESPASYNCHTTPUPDATESERVER_PRETTY
#define ESPASYNCHTTPUPDATESERVER_DEBUG
#define ESPASYNCHTTPUPDATESERVER_SerialOutput=Serial1
#define ESPASYNCHTTPUPDATESERVER_LITTLEFS

void setup(){
    Serial.begin(9600);
     wifi();
......
}
`

**wifi.h**
`
DefaultHeaders::Instance().addHeader("Access-Control-Allow-Origin", "*");
  server.onNotFound(notFound);
  // server.onFileUpload(onUpload);
  updateServer.setup(&server, "/upload_ota3", "admin", "admin");
  server.begin();
  Serial.println("Server begin");
  Serial.println();
`

**Error**
<details><summary>Details</summary>
<p>
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
07:38:07.485 -> 
07:38:07.485 -> Exception (0):
07:38:07.485 -> epc1=0x4021864c epc2=0x00000000 epc3=0x40256d10 excvaddr=0x00000000 depc=0x00000000
07:38:07.576 -> 
07:38:07.576 -> >>>stack>>>
07:38:07.576 -> 
07:38:07.576 -> ctx: sys
07:38:07.610 -> sp: 3fffe9a0 end: 3fffffb0 offset: 0190
07:38:07.610 -> 3fffeb30:  0000000a 000000ff 0000000f 402213fa  
07:38:07.677 -> 3fffeb40:  00008000 00000001 00000001 40100f0f  
07:38:07.744 -> 3fffeb50:  4023cd22 80af1999 000000b8 40100688  
07:38:07.808 -> 3fffeb60:  00000000 00000000 3fffc248 4000050c  
07:38:07.845 -> 3fffeb70:  400043e6 00000030 00000016 ffffffff  
07:38:07.913 -> 3fffeb80:  400044ab 3fffc718 3fffec70 08000000  
07:38:07.913 -> 3fffeb90:  60000200 08000000 00000000 00000000  
07:38:08.007 -> 3fffeba0:  0000ffff 00042035 00002035 000eb000  
07:38:08.045 -> 3fffebb0:  00000001 00000218 00000000 00000030  
07:38:08.079 -> 3fffebc0:  3ffef750 00000000 00000000 3ffef3a4  
07:38:08.111 -> 3fffebd0:  3ffeeeb0 3ffef780 3ffef068 401017df  
07:38:08.177 -> 3fffebe0:  4010560e 3ffef780 3ffef780 3fff1308  
07:38:08.244 -> 3fffebf0:  00007fff 40104f5d 3ffef780 401035ac  
07:38:08.277 -> 3fffec00:  40105511 3ffef780 3ffeeeb0 00040000  
07:38:08.344 -> 3fffec10:  40105511 3ffef780 3ffeeeb0 00040000  
07:38:08.377 -> 3fffec20:  505d0000 00080100 00000000 00000001  
07:38:08.444 -> 3fffec30:  401057e9 00080000 00000000 00000022  
07:38:08.511 -> 3fffec40:  40103ab6 00000000 0000007f 06b0cfbe  
07:38:08.544 -> 3fffec50:  00000000 4000444e 4000050c 3fffc278  
07:38:08.612 -> 3fffec60:  00000002 40106c03 00000001 60000200  
07:38:08.644 -> 3fffec70:  00000002 4000410f 00001001 00000005  

</p>
</details> 

How use example styling code, debug, serial out and use LittleFS?
Because, my code forever reset after upload firmware.

IPdotSetAF, 2024-07-30:
Hi @Sebuahhobi .
- Have you tried running the example sketch that comes with the library? It should probably work on a esp12 module.
- also please enable the exception decoder so we can read the exception message.
To use optional features follow these steps:
1. If using PlatformIO: add build flags tag to your project and depending on the feature you need , add the associated build flag (see the readme of this repo) ,to the build flags list. Rebuild and upload the firmware and it should work.
2. If using arduino IDE: I think you might have to add the build flags globally in the arduino files manually, i am not sure though.

IPdotSetAF, 2024-07-30:
> #define ESPASYNCHTTPUPDATESERVER_PRETTY
> #define ESPASYNCHTTPUPDATESERVER_DEBUG
> #define ESPASYNCHTTPUPDATESERVER_SerialOutput=Serial1
> #define ESPASYNCHTTPUPDATESERVER_LITTLEFS

this way of configuring does not work at the moment.
but I will look into it.

Sebuahhobi, 2024-07-30:
I success change pretty template html. I define on ESPAsyncHTTPUpdateServer.h
But, my code forever restart after click update firmware

IPdotSetAF, 2024-07-30:
please follow the steps below and send me a decoded stacktrace
https://www.donskytech.com/esp8266-exception-decoder/#:~:text=How%20to%20use%20the%20ESPException%20Decoder%3F

closed 2025-03-31 by IPdotSetAF