IPdotSetAF/ESPAsyncHTTPUpdateServer #18: Firmware update OK but Filesystem NOT

plmicia, 2024-09-10:
I am facing difficulties to update FileSystem correctly using this library. Firmware update works well, however when I am trying to upload FileSystem I got "Update error: ERROR[4]: Not Enough Space" or 404. The same *.bin file works flawlesly with classic HTTPUpdate (non Async) and with ElegantOTA (with Async).
Could you double check? Am I missing something?
Above tests were performed in Arduino 1.8.3 and 2.3.2, with 1.2.0. and 2.0.0. library version.

IPdotSetAF, 2024-09-10:
Hey @plmicia 
Could you give me more info of your setup:
What Board, flash size, partition did you use?

plmicia, 2024-09-10:
Sure. Board is ESP8266 D1 mini + 4MB (FS:2MB OTA:~1019KB) flash.
I have double check and in browser I am getting "The connection was reset". ESP8266 is giving reset exception: 

> ..IP Address: 192.168.2.147
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3424, room 16 
> tail 0
> chksum 0x2e
> load 0x3fff20b8, len 40, room 8 
> tail 0
> chksum 0x2b
> csum 0x2b
> v0004dc50
> ~ld
> ����o�s��g|�l�d`b��|s�l�o��n�l`��{�l�l��.......IP Address: 192.168.2.147

Filesystem I am trying to load: [esp8266-FileSystem.zip](https://github.com/user-attachments/files/16948246/esp8266-FileSystem.zip)

IPdotSetAF, 2025-03-31:
hey @plmicia , sorry for the late reply, I had my hands full.

I managed to modify one of my esp1 modules with 4mb flash, and i managed to upload your 1mb fs to it no problem, the problem (if you haven't figured out already by now) is the partitioning of the flash, take a look at this settings:
```yml
board_build.partitions = min_spiffs.csv
board_build.ldscript = eagle.flash.4m2m.ld
```
look [here](https://community.platformio.org/t/two-ways-of-configuring-flash-for-esp8266/39521) for other options

closed 2025-03-31 by IPdotSetAF