IPdotSetAF/ESPAsyncHTTPUpdateServer #19: file sytem buttons removal

mjtrini, 2024-12-10:
Hi,

I want to thank you for your awesome work on the library. For my purposes I don't need the file system to be updated. Can those two buttons be removed or hidden?

Regards,
Michael.

IPdotSetAF, 2024-12-10:
Hey michael.

Thanks for your feedback.
Yes that could be a good option. It could be implemented using build flags so you can disable one of the update modes (either flash or fs).

mjtrini, 2024-12-10:
Good day Mohammad,

Thank you for the rapid response! How exactly do I disable the filesystem update mode?

Regards,
Michael.

IPdotSetAF, 2024-12-10:
@mjtrini to achieve this we need to introduce a new build flag: `ESPASYNCHTTPUPDATESERVER_MODE`.

the behavior will be like this:
- `ESPASYNCHTTPUPDATESERVER_MODE = 0` => Both Update (default)
- `ESPASYNCHTTPUPDATESERVER_MODE = 1` => Only Flash Update
- `ESPASYNCHTTPUPDATESERVER_MODE = 2` => Only FS Update

In this [Code Block](https://github.com/IPdotSetAF/ESPAsyncHTTPUpdateServer/blob/master/src/ESPAsyncHTTPUpdateServer.cpp?plain=1#L29L38), the html content of the web pages are stored, in order to edit them you need to follow these steps:

1. follow the `https://gchq.github.io/....` links above each index. (there are two pages for stylized and non-styled pages)
2. remove the blog corresponding to flash update and copy the resulting array. -> add it in the code with proper conditions
3. remove the blog corresponding to fs update and copy the resulting array. -> add it in the code with proper conditions

now you can switch between them in your project using these build flags in your `platformio.ini`:
```yaml
build_flags = 
	-DESPASYNCHTTPUPDATESERVER_PRETTY
	-DESPASYNCHTTPUPDATESERVER_MODE=1
	;-DESPASYNCHTTPUPDATESERVER_MODE=2
```

Remember to create a fork from this repository and commit your changes into that repo so i can merge your changes into the library properly.

Feel free to ask if there is any uncertainty or question.

GoodLuck

mjtrini, 2024-12-10:
Thanks much!

closed 2025-03-23 by IPdotSetAF