IPdotSetAF/ESPAsyncHTTPUpdateServer #21: Sever side, status of operation

Paolo63ba, 2024-12-30:
Hi,
thanks a lot for your work.

I think it will be a good point, if you include the possibility to know, on server side if the upload is ended correctly (before to do an ESP Restart).

This feature will give the opportunity to display this info before reboot.

Thanks a lot again

Paolo

IPdotSetAF, 2025-03-27:
I could add a few events that you can hook onto to be notified when state change(fs-success, fs-fail, flash-success, flash-fail)

GitHubLionel, 2025-04-27:
Here is my proposition for an `onBeforeUpdate` callback. I add the option to abort the update for any reason.
Header file:
```
#ifndef __ASYNC_HTTP_UPDATE_SERVER_H
#define __ASYNC_HTTP_UPDATE_SERVER_H

#include <ESPAsyncWebServer.h>

typedef void (*ESPAsyncHTTPUpdateServer_cb)(bool *abort, int *code);

class ESPAsyncHTTPUpdateServer
{
public:
    ESPAsyncHTTPUpdateServer();

    void setup(AsyncWebServer *server)
    {
        setup(server, emptyString, emptyString);
    }

    void setup(AsyncWebServer *server, const String &path)
    {
        setup(server, path, emptyString, emptyString);
    }

    void setup(AsyncWebServer *server, const String &username, const String &password)
    {
        setup(server, "/update", username, password);
    }

    void setup(AsyncWebServer *server, const String &path, const String &username, const String &password);

    void updateCredentials(const String &username, const String &password)
    {
        _username = username;
        _password = password;
    }

		void setOnBeforeUpdateCallback(const ESPAsyncHTTPUpdateServer_cb &callback)
		{
			_onBeforeUpdate_cb = callback;
		}

protected:
    void _setUpdaterError();

private:
    AsyncWebServer *_server;
    String _username;
    String _password;
    bool _authenticated;
    String _updaterError;
    bool _updateAborted;
    int _abortCode;
    ESPAsyncHTTPUpdateServer_cb _onBeforeUpdate_cb = NULL;
};

#endif

```
cpp code:
```
#include <Arduino.h>
#include "StreamString.h"
#include "ESPAsyncHTTPUpdateServer.h"

#ifdef ESP32
    #ifdef ESPASYNCHTTPUPDATESERVER_LITTLEFS
        #include <LittleFS.h>
    #else
[... 189 lines cut ...]
                {// true to set the size to the current progress
                    Log("Update Success: \nRebooting...\n");
                }
                else
                    _setUpdaterError();
#ifdef ESPASYNCHTTPUPDATESERVER_DEBUG
                ESPASYNCHTTPUPDATESERVER_SerialOutput.setDebugOutput(false);
#endif
            }
            else
                return;
        });
}

void ESPAsyncHTTPUpdateServer::_setUpdaterError()
{
#ifdef ESPASYNCHTTPUPDATESERVER_DEBUG
    Update.printError(ESPASYNCHTTPUPDATESERVER_SerialOutput);
#endif
    StreamString str;
    Update.printError(str);
    _updaterError = str.c_str();
}

```

IPdotSetAF, 2025-04-27:
So these will be sufficient for the events/callbacks:
 
- OnUpdateBegin (UpdateType updateType, int *abortCode)
- OnUpdateFinished (UpdateType updateType, int resultCode)

UpdateType: enum{ Firmware, FileSystem }
resultCode/abortCode: 
  - 0: success/proceed to update
  - non 0: error/abort update

GitHubLionel, 2025-04-28:
Ok, I try something. See files joined.

[files.zip](https://github.com/user-attachments/files/19937719/files.zip)

IPdotSetAF, 2025-04-28:
thanks @GitHubLionel , you should probably follow the [contribution guideline](https://github.com/IPdotSetAF/ESPAsyncHTTPUpdateServer?tab=readme-ov-file#contribution) .

> You are free to contribute to this project by following these steps:
>   1. Fork this Repo.
>   2. Create a new branch for your feature/bugfix in your forked Repo.
>   3. Commit your changes to the new branch you just made.
>   4. Create a pull request from your branch into the `master` branch of This Repo([https://github.com/IPdotSetAF/ESPAsyncHTTPUpdateServer](https://github.com/IPdotSetAF/ESPAsyncHTTPUpdateServer)).

so you will also be listed in the contributors of the project properly

IPdotSetAF, 2025-04-28:
i just wanted your name to be listed in the repo for the time you put into it😊, but if you don't mind that i will apply all of your changes(task+events) myself don't worry👍😁

closed 2025-05-01 by IPdotSetAF

GitHubLionel, 2025-05-02:
Great :)
I have just some improvements:
- make a compromise: use Ticker only for ESP8266, use native task for ESP32
- _updateType and _updateResult initialized in constructor to avoid compiler warning
- _updateType and _updateResult should be initialised only one time in the `if (!index)` condition
- replace comparison to `emptystring` by `
- set `(void) filename;` to avoid compiler warning

[file.zip](https://github.com/user-attachments/files/20013263/file.zip)

see file with this corrections.

IPdotSetAF, 2025-05-02:
@GitHubLionel , this takes me so much time, follow the contribution guidelines