IPdotSetAF/ESPAsyncHTTPUpdateServer #22: Why is "Update Success! Rebooting..." not displayed?

ivanfd, 2025-02-26:
Hi. 
Why is "Update Success! Rebooting..." not displayed?  After update it shows "IP /update?name=filesystem"

IPdotSetAF, 2025-02-26:
Hey @ivanfd .

The behavior is a little different in esp8266 and esp32. Due to the flashing and rebooting procedure differences.

What board are you using?
In what update modes did you not see the message?

ivanfd, 2025-02-26:
Thanks for the reply.
I am using NODEMCU ESP12E (ESP 8266).
Both Update Firmvare and Update FileSystem do not output a message.

If I use ESP8266HTTPUpdateServer, it outputs after successful firmware. But it does not support ESPAsyncWebServer.

ivanfd, 2025-02-26:
The microcontroller reboots, but does not display "Update Success! Rebooting..." in the web interface

GitHubLionel, 2025-04-27:
I have the same problem and found "a solution" : just remove this lines:
```
            delay(1000);
            ESP.restart();
```
Another point: `send_P` is deprecated replaced by `send`. Same for `beginResponse_P` replaced by `beginResponse`.

GitHubLionel, 2025-04-27:
And another solution I found is to use a task. For example:
```
void reboot_task_function(void*)
{
	for (;;)
	{
		vTaskDelay(pdMS_TO_TICKS(1000));
		ESP.restart();
	}
	vTaskDelete(NULL);
}
```
Then replace 
```
            delay(1000);
            ESP.restart();
```
By `xTaskCreatePinnedToCore(reboot_task_function, NULL, 4096, NULL, 2, NULL, 0);`

IPdotSetAF, 2025-04-27:
thanks for the solution @GitHubLionel .
could you open a pull request, addressing the fixes you mentioned? i'll test and probably publish a version when i get home.
you can follow contribution guideline in the readme.

IPdotSetAF, 2025-04-27:
> And another solution I found is to use a task. For example:
>
> ```
> void reboot_task_function(void*)
> {
> 	for (;;)
> 	{
> 		vTaskDelay(pdMS_TO_TICKS(1000));
> 		ESP.restart();
> 	}
> 	vTaskDelete(NULL);
> }
> ``` [...]

does this solution also need a manual reset?

GitHubLionel, 2025-04-27:
This solution no need a manual reset: the task do the reboot.
Since I am here, I would suggest an improvement. Is it possible to add an `onBeforeUpdate` callback that allow to perform some operations before the update action. It is useful when you want to stop cleanly some operation before the update.

IPdotSetAF, 2025-04-27:
> This solution no need a manual reset: the task do the reboot. 

Thats great.

> Since I am here, I would suggest an improvement. Is it possible to add an onBeforeUpdate callback that allow to perform some operations before the update action. It is useful when you want to stop cleanly some operation before the update.

Take look at #21 . You can request callback functions there👍

GitHubLionel, 2025-04-27:
Yes that's what I need. Just the onBefore because I don't see the goal for an onAfter.
See my proposition on [#21 ](https://github.com/IPdotSetAF/ESPAsyncHTTPUpdateServer/issues/21) with the correction for reboot.

closed 2025-04-28 by IPdotSetAF

GitHubLionel, 2025-04-29:
Hi
I think using a ticker is not a good solution because it requires an external library and probably requires more resources.
Task is native for ESP32 and use a boolean for ESP8266 is very light.