IPdotSetAF/ESPAsyncHTTPUpdateServer #26: updateMode Endpoint

IPdotSetAF, 2025-03-27:
Instead of providing a different template for each updateMode*style, add an endpoint for fetching updateMode from the server and modify the content accordingly.

pros:
- easier development and maintenance
- easier customization
- runtime configurable

cons: 
- higher flash memory usage
- content flickering

wuhgit, 2025-03-27:
In my opinion, a low flash memory usage is paramount, especially for poor flash memory devices like esp01.

But I don't know if most users will agree with me on this.

IPdotSetAF, 2025-03-27:
I agree with you, that was indeed my main goal in developing this library, but after setting up the most minimum project for testing the library, I remembered that we already are unable to update Flash on esp01 due to AsyncWebServer's huge footprint.
so maybe we should move on with that idea and make the library more feature rich/maintainable