IPdotSetAF/ESPAsyncHTTPUpdateServer #31: I personally recommend using uncompressed HTML plain text for the response content

mifarelight, 2025-05-20:
https://github.com/IPdotSetAF/ESPAsyncHTTPUpdateServer/blob/f32cb7adb67b344abcf6363a026bfe96d65f81f9/src/ESPAsyncHTTPUpdateServer.cpp#L35C1-L53C7

like:
```html
<!DOCTYPE html>
<html>
<head>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <title>ESP8266 Web Server</title>
  <style>
    body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }
    .button { background-color: #4CAF50; border: none; color: white; padding: 16px 40px;
              text-decoration: none; font-size: 30px; margin: 2px; cursor: pointer;}
    .button2 {background-color: #555555;}
  </style>
</head>
<body>
  <h1>ESP8266 Web Server</h1>
  <p>GPIO state: <span id="state">%STATE%</span></p>
  <p><a href="/on"><button class="button">ON</button></a></p>
  <p><a href="/off"><button class="button button2">OFF</button></a></p>
  <script>
    setInterval(function() {
      // Call a function repetatively with 2 Second interval
      getData();
    }, 2000); //2000mSeconds update rate
    
    function getData() {
      var xhttp = new XMLHttpRequest();
      xhttp.onreadystatechange = function() {
        if (this.readyState == 4 && this.status == 200) {
          document.getElementById("state").innerHTML = this.responseText;
        }
      };
      xhttp.open("GET", "/state", true);
      xhttp.send();
    }
  </script>
</body>
</html>
```

IPdotSetAF, 2025-05-20:
thanks @mifarelight for the recommendation.

one of our goals is to minimize the flash usage footprint of the library, to achieve it, we decided to go with compressed html content.

we still support multiple ways to customize the html content, either by using the build flags or in case of further customization you can modify the plain HTML files located in the `www` folder. after that u need to run the `codeGenerator.py` script located in scripts folder which will result in the following compressed contents.

> https://github.com/IPdotSetAF/ESPAsyncHTTPUpdateServer/blob/f32cb7adb67b344abcf6363a026bfe96d65f81f9/src/ESPAsyncHTTPUpdateServer.cpp#L35C1-L53C7

mifarelight, 2025-05-27:
I would rather use a script (codeGenerator.py) to generate the compressed file than hard-code the compressed content into the code.